JAndrassy/arduino-firmware-wifilink #1: After put data/config.json, there is no effect.

kalsolio, 2018-01-03:
I add a right config.json in data/config.json. After Upload like this, My uno wifi board has no ip address.
[image]
[image]

kalsolio, 2018-01-03:
I put ssid and password into config.json

kalsolio, 2018-01-03:
After set ssid and password by webpanel, ReUpload SPIFFS also have no effect, I can saw the last setting info, not the setting by config.json file.

kalsolio, 2018-01-03:
I found this.
WiFi.begin(); // connect to AP with credentials remembered by esp sdk

closed 2018-01-03 by kalsolio

jandrassy, 2018-01-03:
sorry, I [changed it](https://github.com/jandrassy/arduino-firmware-wifilink/commit/36d8ce250ae4575e27f738c2117f1a1efcc16625#diff-04c6e90faac2675aa89e2176d2eec7d8) recently