JAndrassy/arduino-firmware-wifilink #2: Can't upload firmware to arduino Uno Wifi

wcpettus, 2018-04-02:
I'm trying to switch to your version of the WiFiLink library and figured I should update to your firmware first.  When I get to the [upload step](https://github.com/jandrassy/arduino-firmware-wifilink#upload), I get this return:
```
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
```

Using an Arduino Uno WiFi developer edition, I've never done a firmware update before, so possibly I don't have it in the right mode for flashing (I was trying to use the ESP B/L button, holding it while reconnecting).  It wasn't clear from your test setup example how to extrapolate to this board with a built-in chip.

jandrassy, 2018-04-02:
[UNO WiFi with WiFi Link firmware](https://github.com/jandrassy/UnoWiFiDevEdSerial1#uno-wifi-with-wifi-link-firmware)

wcpettus, 2018-04-03:
Thanks, I think that worked.

If I use the CheckWiFiLinkFirmwareVersion sketch from the WiFiLink library, it shows 1.1.0, the last tag was 1.0.1, so that's odd.  But I am getting much better behavior in terms of disconnecting and reconnecting a telnet, which was my goal.

jandrassy, 2018-04-03:
I changed the version number in my repositories, because one bug fix required changes on booth sides (firmware and library) and I added a new function WiFi.resetEsp(). (but with Uno WiFi use better Serial1.resetESP()). Everything Uno WiFi related is documented in the Uno WiFi Serial1 project. See examples of Uno WiFi Serial1 library. WiFi Link docs are for common use with any Arduino+esp8266.

closed 2018-04-03 by jandrassy