JAndrassy/arduino-firmware-wifilink #6: ESP-03

jfmateos, 2019-04-01:
Dear Jandrassy,

Could I use this project with an ESP-03? It only has 512Kbytes, so the max SPIFF size is 128Kbytes, but I read that the ESP8266 filesystem uploader needs at least 512kB.

Thanks and kind regards from madrid

jandrassy, 2019-04-01:
try to use it without SPIFFS. set SSID and password in sketch

jfmateos, 2019-04-01:
Thanks, You put me on the way.
I have managed to get port 192.168.1.140 in the Arduino IDE.
I have cofigured the board as "Arduino Uno WiFi" (in fact it is a bare atmega328p with minicore bootloader).
I have connected:
RX from ESP-03 to TX on Atmeta328p
TX from ESP-03 to RX on Atmeta328p
GPIO0 from ESP-03 to Reset on Atmega 328p

But I get the following error:
https://i.paste.pics/6096385882ec9b4a1a4380bd86fac892.png

jfmateos, 2019-04-01:
The minicore in the Atmega328p is configured with an external 8MHz clock

jandrassy, 2019-04-01:
you have Optiboot with custom frequency and baud rate in the ATmega?
try to set DEFAULT_BAUD in esp8266-serial.c in dfu library

jfmateos, 2019-04-01:
Nice it worked. Thank you very much Jandrassy.

https://i.paste.pics/fbf514580dbf7b6924a88a1cbf3e9ff6.png

The baud rate for minicore 328p 8MHz was 38400.

Nevertheless, I have found esp-03 boards with 1MB... so I will try again with SPIFFS.

jandrassy, 2019-04-01:
btw I am [Juraj](https://github.com/jandrassy). jandrassy is only a login created from first letter of the first name and the second name.