JAndrassy/arduino-firmware-wifilink #7: WiFiLink and dfu fails to burn a sketch from ESP8266 to Arduino Nano ?

SrikanthBala, 2019-07-21:
I have successfully uploaded a sketch via WiFi Link from a generic ESP8266 12E (Node MCU 1.0) to Arduino Uno. However I tried the same on the Arduino Nano and it failed and probably destroyed my bootloader
Is there some way around ?

jandrassy, 2019-07-21:
dfu uses 115200 baud for Optiboot. older Nano has a bootloader which requires slower baud rate. Change the bootloader to 'new' Nano bootloader or better burn the bootloader as for Uno and use the Nano as Uno.
(or try to change the baud rate in dfu library. but that I didn't try)

SrikanthBala, 2019-07-22:
I guessed as much and was successful after burning the Nano with a UNO bootloader as in the current version they are the same. Thanks a lot for your prompt reply. Sorry I did not close the issue as soon as I was successful
Thanks for the great support. My upload is soooooo fast now !

closed 2019-07-22 by SrikanthBala