JAndrassy/arduino-firmware-wifilink #8: Failed to reset the board, please reset maually error

SrikanthBala, 2019-08-28:
First of all, thank you very much for the wifilink.  It has saved me a lot of trouble flashing my remotely placed device.

However I have a small nagging problem.....

My setup uses a NodeMCU V1.0  to upload sketch to a Arduino Nano with the UNO  optiboot bootloader flashed.

The system works perfectly and the ESP is able to successfully upload the sketch to the nano with no problems and the sketch works too. 
However at the end of the flashing, it gives an error "Failed to reset the board, please reset maually" error, which is nagging me.
The Arduino Nano RST pin is connected to ESP8266  GPIO 05 (Pin D1 of Node MCU) as defined "esp8266-serial-arduinouno-hacked.cpp" 

Though everything works, why the error ?  3.3V to 5V voltage level is unlikely to be the problem as the board resets correctly otherwise the upload would not have occurred.

Is the flasher unable to detect the reset of the Arduino Nano ? If so what is the solution   ?
Thanking you

jandrassy, 2019-08-28:
without working reset the Nano could not be flashed, because the process starts with reset. so it is only some communication glitch in the confirmation of the reset. which version of the OTA upload tool in IDE do you have? it is installed with Arduino boards packages for AVR and SAMD. In console verbose mode you should see the upload command with the version of the tool.

jandrassy, 2019-08-28:
it is possible that the Nano already resets the esp8266 while the upload tool waits for notification about the reset. try to add a delay in setup() of the Nano sketch before init() of the library

SrikanthBala, 2019-08-28:
I set the upload output option to verbose in preferences but it does'nt  show a version number. It is the latest arduino ide 1.8.9 and I have updated all boards 'updateable ' and tried again. Thi sis teh output :

Global variables use 843 bytes (41%) of dynamic memory, leaving 1205 bytes for local variables. Maximum is 2048 bytes.
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/arduinoOTA -address 192.168.1.13 -port 80 -sketch C:\Users\xxxxxxxxxxxxxxxxxx\arduino_build_886809/Trialxxxxxxxx.ino.hex -upload /pgm/upload -sync /pgm/sync -reset /log/reset -sync_exp 204:SYNC 
Resetting the board
Waiting for the upload to start
SYNC
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done

Sketch uploaded successfully
Resetting the board
Failed to reset the board, please reset maually

SrikanthBala, 2019-08-29:
I have not connected any Nano pin to the ESP8266. How can the nano reset the ESP then ? Am I supposed to connect any Nano pin to the ESP ? 
The software for the ESP itself has a ESP ArduinoFirmwareESP.ino and dfu seem to have some resetting of the ESP via software. I'll see if there is a delay I can incorporate there

jandrassy, 2019-08-29:
I assumed your sketch is with WiFiLink library based on examples of the library. The examples ask esp8266 to reset itself with `WiFi.resetESP();`

The truth is my patch of the dfu library to work with the default Arduino IDE upload tool is a big compromise. Originally the dfu library expects a different upload tool. [Here is it described](https://github.com/jandrassy/arduino-firmware-wifilink/blob/0545636df8defc15f64e7e7d264e9745ae6de32e/README.md#sketch-ota-upload-tool) in an older version of the README.

SrikanthBala, 2019-08-29:
So, do I need to download the other  tool ? Currently if it is working as is , I would not like to patch anything or change too much.  as if something goes wrong, I'll not be able to complete the project in time.

closed 2019-09-01 by SrikanthBala