JAndrassy/arduino-firmware-wifilink #9: Can't open webpage.

suraj022, 2019-10-29:
I'm trying to flash the firmware on a generic nodemcu board using the "UNOWIFIDEVED" parameters
firmware flashes successfully and SSID appears (and connects). I've also uploaded sketch data folder using "esp8266 sketch data uploader" tool.
But the webpage doesn't seem to open by pointing my browser to 192.168.240.1
Am I missing anything?

jandrassy, 2019-10-29:
please try 192.168.4.1

suraj022, 2019-10-29:
@jandrassy tried it without any success. Anyways, default IP in the code is set to 192.168.240.1

Also I noticed just now that my android phone fails to obtain IP address from the AP. which means DHCP might not be running?

jandrassy, 2019-10-29:
try to flash the firmware with Tools menu option "Erase flash: Sketch + WiFi Settings"

suraj022, 2019-10-29:
Did that aswell. no change. webpage still wont load.

jandrassy, 2019-10-29:
I found the error. Fix is committed. It was an old bug by arduino.org coders. I should have noticed it much earlier.

suraj022, 2019-10-30:
Thanks. but it still wont load. I can't even ping to the module.
[image]

jandrassy, 2019-10-30:
try to flash some ESP8266 example for example  BasicHttpClient to connect to your AP and let the esp8266 store the ssid and password. Then upload the WiFiLink firmware and it should connect to AP and be accessible in that network

suraj022, 2019-10-31:
> I hope you didn't select Arduino/Uno WiFi to upload the firmware. it would not work on NodeMcu

yes ofcourse. the  wouldn't compile on "Arduino/Uno WiFi". anyways. Connecting nodemcu to a AP is not an option here As I plan to use it only in AP mode.

suraj022, 2019-11-01:
> try to flash some ESP8266 example for example BasicHttpClient to connect to your AP and let the esp8266 store the ssid and password. Then upload the WiFiLink firmware and it should connect to AP and be accessible in that network

Just did that. and it does work when connect to  an access point( i.e. in STA mode)

jandrassy, 2019-11-01:
and SoftAP still doesn't work?

suraj022, 2019-11-01:
> and SoftAP still doesn't work?

No.

jandrassy, 2019-11-01:
could you try to switch modes in WebPanel? to STA and the back to AP+STA

suraj022, 2019-11-01:
> could you try to switch modes in WebPanel? to STA and then back to AP+STA

Yes. works in this way in AP+STA mode. but not when I freshly upload the code.

jandrassy, 2019-11-01:
and after fresh upload now is it available as STA?

suraj022, 2019-11-02:
> and after fresh upload now with flash erase is it available as STA?

Yes works with STA mode.

jandrassy, 2019-11-02:
AP is not started if esp8266 has stored STA mode
https://github.com/jandrassy/arduino-firmware-wifilink/blob/c634eca7f34a117f0f3a8167f482636c74c616d1/ArduinoFirmwareEsp/ArduinoFirmwareEsp.ino#L110

suraj022, 2019-11-04:
I've found a potential fix. by commenting out the `initHostname();` function in `void setup()` the code works normally in both STA mode and AP mode.

closed 2020-05-17 by jandrassy