JAndrassy/arduino-firmware-wifilink #12: Uno WiFi Dev Edition

Damianoux, 2020-06-13:
HI,
I have a problem with Arduino Uno Wifi Dev Edition (Rev1).
In installed the firmware and everything works, except for te fact that mDNS doen not start on boot.
I cannot discover until I change the hostname in the web panel.
After doing this the device is discoverable and I can program OTA.
On the next reboot mDNS does not work  and I need repeat the procedure to program once again.

jandrassy, 2020-06-13:
Hi. what version of esp8266 Arduino did you use to build the firmware?

Damianoux, 2020-06-13:
Hi,
Should be 2.7.1.

jandrassy, 2020-06-14:
try in setup() of the firmware disabling the MDNS start by OTA library
`ArduinoOTA.begin(false);`
to not be in conflict by MDNS initialised by the firmware sketch

Damianoux, 2020-06-14:
Hi jandrassy,
No it didn't work by disabling ArduinoOTA.begin in setup(), but maybe I found a different arrangement. It seems that setup() tries to initializa mDNS before WIFI has started.
So I moved all initialization in loop(), checking that WiFi.status() is connected before.
If you are interested in PR i will test it for a while than i will push changes on my fork.

jandrassy, 2020-06-14:
this https://github.com/jandrassy/arduino-firmware-wifilink/commit/401af1d75965b3dc469dfc4c6ccae07965a48c31 ?

Damianoux, 2020-06-14:
Didn't try this.
I delay all initializations until wifi is available. Seems To work for Now. If It does Not have any Side effect in The Next few days i Will push The changes.
But Now after powering The board i immediately Can See The wifi Port.

jandrassy, 2020-06-14:
setWiFiConfig() does the WiFi setup and waits for the connection

Damianoux, 2020-06-15:
So moving initMDNS after setWiFiConfig should start mdns after the connection has been estabilished?
I will try this as soon as I have the time.

closed 2022-02-09 by jandrassy