JAndrassy/arduino-firmware-wifilink #13: mDNS in STA mode

plop3, 2020-07-10:
mDNS doesn't work in STA mode.
On the AP network, it's work fine (I can see the host in IDE Adruino).
AP:
ping arduino.local -> OK
STA:
mDNS is not propagated on the local network (STA)
ping arduino.local -> KO and i can't see the host in IDE Arduino.

jandrassy, 2020-07-10:
hi. it is a problem of esp8266 mdns library from 2.6.3 and should be fixed from 2.7.0 (to not working on AP in AP+STA mode)

plop3, 2020-07-10:
Hello,
I have 2.7.2 ESP8266 version (2.7.1 tested too).
I have modified the setup procedure for test:
- Modified the functions order,
- Remove initMDNS();
- Add ArduinoOTA.setHostname("myesp8266");

And it's work.
I can access at "myesp8266.local" in Arduino IDE.
It's quick and dirty, but it's work :-)
...
  digitalWrite(WIFI_LED, LOW);
  //initMDNS(); 
  CommunicationLogic.begin();
  SPIFFS.begin();
  initHostname();
  setWiFiConfig();
  ArduinoOTA.setHostname("myesp8266");
  ArduinoOTA.begin();             //OTA ESP
  initWebServer();                 //UI begin
...

plop3, 2020-07-10:
I just tested the upload to the Arduino Nano (Optiboot) and it works.
I just have a message:
   "Sketch uploaded successfully
    Resetting the board
    Failed to reset the board, please reset maually "

But the reset is done.

closed 2022-02-09 by jandrassy