JAndrassy/arduino-firmware-wifilink #14: Can't Open Web Page

archiebaldy, 2021-12-30:
I have followed all instructions to the letter.
Used ArduinoJson 5.35, verfied all files, uploaded to NodeMCU ESP8266 with settings as described.  Then uploaded the webpanel data files.  All appears to work correctly, then I try to connect to the AP.  Works on PC, but an Android phone cannot receive an IP address.
When I try to access the pages by IP 192.168.240.1 the page cannot be found.
I have worked my way through all the issues in this git, both "resolved" and open but none have resolved this issue.

Any ideas please?

jandrassy, 2021-12-30:
what version of esp8266 arduino do you use? if 3.0.2, then try 2.7.4. if 2.7.4 then try 3.0.2

archiebaldy, 2021-12-31:
Thanks for replying so quickly.
I have currently been running ESP8266 Community v 3.0.2 under Arduino IDE 1.8.19
As per your advice, I removed ESP8266 completely then re-installed v 2.7.4, then re-started the IDE (just in case)
Sketch verified okay, uploaded to ESP with "All Flash" enabled, re-uploaded the data files, no problem just as before.
Connected to the ESP via wifi - Network properties on the PC showed I am connected to DNS server 192.168.240.1 under IP address 192.168.240.100

With this config, I could now connect and web pages were visible both from a Windows PC and an Android phone. SUCCESS!!

Just for troubleshooting, I went back and updated ESP8266 Community to 3.0.2 and repeated the process. 
Sketch verification, uploading all went normal.  I could connect to the AP as before but was not expecting any web page.

Now it gets weird.

Unlike yesterday under 3.0.2, today via a Windows PC, network properties report the same DNS server and IP address connection as seen with ESP8266 2.7.4
The Windows PC was still able to connect to the ESP8266 and display web content.  Don't understand why.
Thinking that the PC and browser might have something in cache, I rebooted the PC then tried connecting with another browser, but it still connected and displayed the web content correctly.

But, under 3.0.2 the Android phone gets stuck at Obtaining IP address with no further access possible.

To test repeatability I now went back and repeated the replacement of 3.0.2 with 2.7.4 and now both the PC and phone could connect with no apparent issues.

One small difference I noted under the two scenarios.  On the first page of the wifilink pages "Overview"
With 3.0.2 Hostname was "ESP-CEE28B" (default name)
With 2.7.4 Hostname was "arduino" (nothing that I changed)

As you probably guessed I am not a programmer, just a tinkerer - thanks for the feedback and assistance, I hope my test results help.

Question - is there any way to update wifilink to work under 3.0.2, if not I'll add header notes to the sketches to remind myself in 6 months that it need 2.7.4.

Thanks again.

jandrassy, 2021-12-31:
I think very few people use WiFiLink these days. What is you reason to use WiFiLink? Uno OTA?

closed 2021-12-31 by jandrassy

archiebaldy, 2021-12-31:
A very simple thing - I have an oscilloscope tube clock that I built from a kit originally used a DCF77 time signal, but I am not in Europe so I installed an ESP8266 to enable NTP time synchronization via my home network.  
Now I am getting older and plan to pass it on to a non-tech family member so I want to make an easy portal for them to log into the device and change the SSID and password for themselves, also a parameter for their time zone.

I have been searching for something that allows for this for a while and up to now wifilink seems to be the best solution  with a couple of mods in the web pages for additional parameters.  If you know of something that will achieve a similar result, I'd love to know.

jandrassy, 2021-12-31:
for esp8266 the WiFiManager library is common for WiFi provisioning. so if you don't use the WiFiLink library but write your own esp sketch then WiFiManager is the right way.

I have an alarm clock project with esp8266 only which once a day connects to WiFi and retrieves NTP time, then turns off the WiFi. for configuration SoftAP is started in menu (or if in setup( the WiFi connection fails.)
[https://github.com/jandrassy/SunriseClock](https://github.com/jandrassy/SunriseClock)

archiebaldy, 2021-12-31:
Fantastic! Thanks so much - I will study your alarm clock and see how much I can learn from it.

Happy New Year to you!

jandrassy, 2021-12-31:
there is one 'wild' part in the project. the WiFiManager doesn't support dropdown list selection so my custom parameters are a little complicated.