JAndrassy/arduino-firmware-wifilink #15: ArduinoJson 6 updated needed

w1r3d90d, 2022-01-23:
ArduinoJson 6 updated needed
That's all

jandrassy, 2022-01-24:
Hi. do you use this firmware? Why?

w1r3d90d, 2022-01-25:
Well I was trying to.  I need ArduinoOTA ability with a standard Arduino Uno.  I have plenty of esp8266-01 boards and this project seemed to fit the need perfectly.  I was able to get around the JSON issue but I now cannot get the SPIFFS data to load.  I think the esp-01 doesn’t have enough flash for the project binary and SPIFFS to fit.

jandrassy, 2022-01-25:
if you use Linux, you can use this to upload to Uno with esp8266
https://github.com/jandrassy/lab/blob/master/AvrDudeTelnet/AvrDudeTelnet.ino

or you could try
https://github.com/jeelabs/esp-link

w1r3d90d, 2022-01-25:
Thanks
I could probably use either of those.  
Thanks